paviro/MMM-Facial-Recognition-Tools #1: train.py crashes

PtrBld, 2016-04-23:
Hi,
when I try to start the training for my images I choose the "LBPHF (recommended)" option after that it reads my images but chrashes with: 
`Training model...
Traceback (most recent call last):
  File "train.py", line 104, in <module>
    model = cv2.createLBPHFaceRecognizer()
AttributeError: 'module' object has no attribute 'createLBPHFaceRecognizer'`

Am I missing dependencies?

paviro, 2016-04-23:
Hi,
Are you sure you installed all the opencv2 dependencies? If so try installing a newer version from source.

Cheers,
Paul

paviro, 2016-04-23:
Are you using open cv3? If yes see http://answers.opencv.org/question/56859/opencv-300-facerecognizer-and-python-bindings/

PtrBld, 2016-04-25:
You were right. I had to compile a newer version by myself.
Thank you!

closed 2016-04-25 by PtrBld

paviro, 2016-04-25:
Glad you could solve it! Could you may add something to the README in case someone else has the same problem?

PtrBld, 2016-04-30:
Ok after spending some hours investigating I came to the result that I have to buy a new pi because rpi b+ is not fast enough... So I do not recommend to use this module with the rpi b+ at all. And therefore noone should need to go through the same problems as I did.

paviro, 2016-04-30:
Does it work if you set a higher timeout for the recognition interval?

PtrBld, 2016-04-30:
No it is a Problem with the open cv library itself

paviro, 2016-04-30:
Ok.