paviro/MMM-Facial-Recognition-Tools #2: raspberry pi3 error

vanimesh76, 2016-04-25:
traceback (most recent call last):
  File "capture.py", line 18, in <module>
    capture.capture()
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/capture.py", line 40, in capture
    camera = config.get_camera()
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/config.py", line 32, in get_camera
    return webcam.OpenCVCapture(device_id=0)
  File "/home/pi/Downloads/MMM-Facial-Recognition-Tools-master/facetrainer/lib/webcam.py", line 29, in __init__
    self._camera.open()
TypeError: Required argument 'device' (pos 1) not found

paviro, 2016-04-25:
Sorry, the facetrainer currently does not work on the Raspberry Pi! Please use your computer.

closed 2016-04-25 by paviro

vanimesh76, 2016-04-25:
ok so then after that i can use the xml file created on the raspberry?

paviro, 2016-04-25:
Yes, exactly 😊 I will add Pi support again some time but you should probably do it on a computer anyway since it is a lot faster.