paviro/MMM-Facial-Recognition-Tools #3: Training set error On laptop

vanimesh76, 2016-04-25:
I have installed open cv3
line 18 error is occurring again.
Not getting anything online for these errors. :(

`Capturing positive training images.
Type c (and press enter) to capture an image.
Press Ctrl-C to quit.
Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    capture.capture()
  File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 56, in capture
    if is_letter_input('c'):
  File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 24, in is_letter_input
    if select.select([sys.stdin, ], [], [], 0.0)[0]:
select.error: (10038, 'An operation was attempted on something that is not a socket')
`

paviro, 2016-04-25:
Please use open cv2 or change the things mentioned [here](https://github.com/paviro/MMM-Facial-Recognition-Tools/issues/1). The other problem should be solved now.

closed 2016-04-25 by paviro

vanimesh76, 2016-04-25:
I re-installed opencv2 

- Now I got my train.py working

- but capture.py is showing the same error.
 

Images will be placed in ./training_data/d

--------------------
Starting process...

Capturing positive training images.
Type c (and press enter) to capture an image.
Press Ctrl-C to quit.
Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    capture.capture()
  File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 56, in capture
    if is_letter_input('c'):
  File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 24, in is_letter_input
    if select.select([sys.stdin, ], [], [], 0.0)[0]:
select.error: (10038, 'An operation was attempted on something that is not a socket')

paviro, 2016-04-26:
Please update your code from this repo!