paviro/MMM-Facial-Recognition-Tools #4: Image capture error

vanimesh76, 2016-04-25:
I re-installed opencv2

- Now I got my train.py working
- but capture.py is showing the same error.

**what am I doing wrong, also if I capture images manually and save it in a folder will the train.py still work?**

Images will be placed in ./training_data/d

Starting process...

Capturing positive training images.
Type c (and press enter) to capture an image.
Press Ctrl-C to quit.
Traceback (most recent call last):
File "capture.py", line 18, in 
capture.capture()
File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 56, in capture
if is_letter_input('c'):
File "C:\Users\Animesh\Downloads\MMM-Facial-Recognition-Tools-master\facetrainer\lib\capture.py", line 24, in is_letter_input
if select.select([sys.stdin, ], [], [], 0.0)[0]:
**_select.error: (10038, 'An operation was attempted on something that is not a socket')_**

vanimesh76, 2016-04-25:
In the facerecognition.py what changes are to be made in "personen = []"

> In `tools/facerecognition_tester/lib/config.py` add add your training labels from `tools/facetrainer` under `personen = []` and specify the used algorithm

paviro, 2016-04-26:
You are still using the old version of the capture code.

paviro, 2016-04-26:
Those that you state. The array with the people, you'll a get it from the training script and the algorithm you choose in the training script.

paviro, 2016-04-26:
But make sure you only change the config! Also note, that that tool is only for testing.

paviro, 2016-04-26:
You can capture images manually. The capture script can convert them for you (it asks at the beginning)

vanimesh76, 2016-04-26:
Thanks now the capture and train.py both are working on windows as well as the Raspberry pi :)

vanimesh76, 2016-04-26:
it is giving error with opencv on both the platforms..
on searching for the error websites suggest it is because the images used are of BGR rather than rgb, but i took those images from two different cameras. stilll.
should I get opencv updtaed to 3.0, also 
python facerecognition.py
Loading training data...
ALGORITHM: LBPH
Training data loaded!
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 34, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

vanimesh76, 2016-04-26:
link to error solution

https://www.google.co.in/#q=p:3737:+error:+(-215)+scn+%3D%3D+3+%7C%7C+scn+%3D%3D+4+in+function+cvtColor

paviro, 2016-04-26:
How did you take the pictures? With the capture script or with another program?

vanimesh76, 2016-04-26:
I took pictures from pi camera and converted them using capture.py, got the .pgm format files, ran the train.py script produced the "training.xml" file.

didi this both on Laptop and Rpi.
but the error is same.

paviro, 2016-04-26:
Can you try to remove all images but one and use train.py again? Have you ever tried to capture them with the capture script and not just convert them on your Windows machine?

vanimesh76, 2016-04-26:
Ok I captured images using the capture.py script successfully on the laptop
tried running the facerecognition script still no,
also did training with single image.

paviro, 2016-04-26:
Just to be clear training worked now?

vanimesh76, 2016-04-26:
yes training worked perfectly well

vanimesh76, 2016-04-26:
on both platforms Rpi and Windows

paviro, 2016-04-26:
Alright glad to hear :)

closed 2016-04-30 by paviro

bikerboymohit, 2016-06-09:
@vanimesh76 & @paviro,

Can you please elaborate how you got passed through the error. I am facing a similar one and finding a hard time passing through it.

Appreciate your time and effort.

paviro, 2016-06-09:
Please post your console output.

bikerboymohit, 2016-06-09:
The mirror does not load when I include the face recognition code into the config.js. Without this code the mirror works perfectly well.
Also, this is the output  I get after I run 

OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737

bikerboymohit, 2016-06-09:
The console does not load when I have the face recognition code in the config.js file. Without this code the mirror works perfectly well.

Also here is the output I get after I run

pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools/facerecognition_tester $ ls
facerecognition.py  lib  README.md  training.xml
pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools/facerecognition_tester $ python facerecognition.py 
Loading training data...
ALGORITHM: LBPH
Training data loaded!
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 42, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

bikerboymohit, 2016-06-09:
I have taken the pictures manually using raspistill and converted them using the capture.py and then ran the train.py to create a training.xml file.

paviro, 2016-06-09:
Training worked? The error only appears on usage?

bikerboymohit, 2016-06-09:
training worked fine. No errors there. The error shows up only after I run the facerecognition.py

bikerboymohit, 2016-06-09:
This is my config.js. But I think the error lies around the facerecognition.py,

{
    module: 'MMM-Facial-Recognition',
    position: 'top_left',
    config: {
        // 1=LBPH | 2=Fisher | 3=Eigen
        recognitionAlgorithm: 1,
        // Threshold for the confidence of a recognized face before it's considered a
        // positive match.  Confidence values below this threshold will be considered
        // a positive match because the lower the confidence value, or distance, the
        // more confident the algorithm is that the face was correctly detected.
        lbphThreshold: 50,
        fisherThreshold: 250,
        eigenThreshold: 3000,
        // force the use of a usb webcam on raspberry pi (on other platforms this is always true automatically)
        useUSBCam: false,
        // Path to your training xml
        trainingFile: 'modules/MMM-Facial-Recognition/training.xml',
        // recognition intervall in seconds (smaller number = faster but CPU intens!)
        interval: 2,
        // Logout delay after last recognition so that a user does not get instantly logged out if he turns away from the mirror for a few seconds
        logoutDelay: 15,
        // Array with usernames (copy and paste from training script)
        users: ['bikerboy'],
        //Module set used for strangers and if no user is detected
        defaultClass: "default",
        //Set of modules which should be shown for every user
        forAllClass: "for_all"
    }
}

bikerboymohit, 2016-06-09:
Any help here is appreciated.

Thanks!

SvenSommer, 2016-06-09:
I deleted all comments in the config.js file within the facereg module section. This solved my error. 
Maybe this can help you also.

bikerboymohit, 2016-06-09:
But I error appears before I can even load up the console. The error
appears when I try to run the facerecognition.py

bikerboymohit, 2016-06-10:
Deleting the comments did help. Thank you.

paviro, 2016-06-10:
Strange..

paviro, 2016-06-10:
Oh and guys! Only add to the config what you want to change not everything that is there!

SvenSommer, 2016-06-13:
Hey vadimnyc,
it seems opencv has trouble converting your images into grey colors. 
But I think this isn't the main problem, neither is it caused by your images captured by phone. I would assume the function cv2.cvtColor isn't getting any images, because the image wasn't loaded properly. There is no exception build in right now, which shouts out and is able to handle an error if the image is NULL.

paviro, 2016-06-13:
Yeah, I need to add a check for empty images. Could you try adding it because I am busy with school at the moment.. :/