paviro/MMM-Facial-Recognition-Tools #5: Facerecognision tester crash

Try2Fly, 2016-05-02:
Hi,

I get the following error as soon as a face is recognized in the tester module:

Traceback (most recent call last):
  File "/home/pi/MMM-Facial-Recognition-Tools-master/facerecognition_tester/facerecognition.py", line 67, in <module>
    cv2.putText(frame, config.personen[label - 1], (x - 3, y - 8), cv2.FONT_HERSHEY_SIMPLEX, 1.5, (255, 255, 255), 1)
IndexError: list index out of range

paviro, 2016-05-02:
Have you added the array with the names to the config?

Try2Fly, 2016-05-02:
No, forgot that. It is working now. But only with a webcam, not with the pi cam module. Is this possible with the pi cam module?

Thanks!

paviro, 2016-05-02:
Should be. What happens when you execute `import picam` in a Python shell?

Try2Fly, 2016-05-02:
ImportError: no module named picam

paviro, 2016-05-02:
You are really running this on a Pi? Have you enabled the cam with raspi-config? Do you have the latest distro?

Try2Fly, 2016-05-02:
seems like my pi camera stopped working. Cannot even use raspistill command to take single picture. Need to get the camera to work again first. If that works, I'll try this again. Thanks a lot!

paviro, 2016-05-02:
Alright, good luck!

Try2Fly, 2016-05-18:
Hi Paviro,

Turned out the camera connection on my pi was broken. Replaced it and the raspberry camera module now works. I was able to create a training file. However when starting the tester I get the following error:

Traceback (most recent call last):
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools-master/facerecognition_tester/facerecognition.py", line 42, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

Can you help me?

ErikTh, 2016-06-15:
Just ran into the same issue. As I see it the facerecognition.py does not switch according to the camera type.

In facerecognition.py:
Line 37 replace
`capture = cv2.VideoCapture(0)`
with
`camera = config.get_camera()`

Line 41 replace
`ret, frame = capture.read()`
with
`frame = camera.read()`

paviro, 2016-11-15:
Could you send a pull request?

paviro, 2016-11-20:
Added thanks :)

closed 2016-11-20 by paviro

YaarPatandarAA, 2016-11-27:
Still having issues with 

File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools-master/facerecognition_tester/facerecognition.py", line 42, in 
image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

YaarPatandarAA, 2016-11-27:
Still having similar issues:

```
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /home/pi/opencv-2.4.9/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 42, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /home/pi/opencv-2.4.9/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor
```

Saltgurkan, 2016-11-27:
Also experiencing issue with PiCam:

```
pi@raspberrypi:~/MMM-Facial-Recognition-Tools/facerecognition_tester $ python facerecognition.py
Loading training data...
ALGORITHM: LBPH
Training data loaded!
--------------------
Picam ausgewählt...
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 45, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor
```

Patrick-Remy, 2016-12-28:
Same here, opened a new issue #11