paviro/MMM-Facial-Recognition-Tools #7: error setting pixformat

tommibaer, 2016-09-03:
I got the following mistake after execute ther 4. step of the recognition-test-tool introduction

_4. Enter the tools/facerecognition_tester directory from a terminal and run python facerecognition.py._

```
pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools/facerecognition_tester $ python facerecognition.py
Loading training data...
ALGORITHM: LBPH
Training data loaded!
libv4l2: error setting pixformat: Device or resource busy
HIGHGUI ERROR: libv4l unable to ioctl S_FMT
libv4l2: error setting pixformat: Device or resource busy
libv4l1: error setting pixformat: Device or resource busy
HIGHGUI ERROR: libv4l unable to ioctl VIDIOCSPICT

OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 42, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

```

Do somebody know why?

paviro, 2016-09-03:
Please check http://forum.magicmirror.builders and come back if you are sure it is a bug :) I don't have a working mirror right now so I can't check :/

GHLasse, 2016-09-13:
I get the same error. :-/

swathi-ssunder, 2016-10-08:
@tommibaer , @GHLasse  - Does anyone of you know how to proceed after this error?  I get the same error. 
@paviro - I get this error even when I run `python facerecognition_tester` without Magic Mirror. Could you please help?

swathi-ssunder, 2016-10-08:
I noticed that this error was only in the `tester`. When I used the module in the Magic Mirror, it seemed to work fine.

paviro, 2016-11-15:
Same for the others? Only in the tester?

paviro, 2016-11-20:
This maybe fixed now could you try?

paviro, 2016-11-20:
Please reopen if not! :)

closed 2016-11-20 by paviro