paviro/MMM-Facial-Recognition-Tools #8: Cant recognize my camera.

Labzz, 2016-10-29:
[image]

Hi Paviro,

I've some errors in the with my raspberry pi installing the tools. I've an USB and a Picam. But for both the same error. 

Are you able to help me? Or do you have some suggestions?
Python isn't my best language ^_^  

greets

stantang986, 2016-10-29:
I got same errors for my usb camera

paviro, 2016-11-15:
Have you tried doing what it says? Entering raspi-config and enabling it?

Labzz, 2016-11-15:
Yes sure!

Labzz, 2016-11-15:
I've done the installation multiple times, the reason in this Picture was the permissions. After the reinstall i clone the git Without sudo. That fixed the first lines.

Labzz, 2016-11-15:
If i'm running the command : camera.capture in the terminal to test, it doesn't work neither ._.

Labzz, 2016-11-15:
Errors still exist, busy for 2/3 weeks already more than 30 hours tryed to fix it haha. But worth it, it is a great feature to add :)

Labzz, 2016-11-15:
http://imgur.com/a/kxxb1 
I can access the camera board on /dev/video0 by running the command:

sudo modprobe bcm2835-v4l2

Labzz, 2016-11-17:
Fixed it, can be closed.

paviro, 2016-11-17:
What was the problem?

Labzz, 2016-11-17:
I wish i could tell you that, i fixed it for myself with a work around by using the convert (2) option> explained in another [(post.https://github.com/paviro/MMM-Facial-Recognition-Tools/issues/9)]

But i'm already fighting with a new error

underline not related on this issue
--------------------------------------------

@raspberrypi:~/MMM-Facial-Recognition-Tools/facerecognition_tester $ python facerecognition.py
Loading training data...
ALGORITHM: LBPH
Training data loaded!
Traceback (most recent call last):
  File "facerecognition.py", line 63, in <module>
    cv2.putText(frame, config.personen[label - 1], (x - 3, y - 8), cv2.FONT_HER$
IndexError: list index out of range

paviro, 2016-11-17:
Have you added the names correctly?

Labzz, 2016-11-17:
i've only added my own name? do i also have to add the default names? 
The following folders exist in training_data 
jd  negative  othername  vincent

my array looks like this: 

personen = ["vincent"]

# Edit the values below to configure the training and usage of the
# face recognition box.
RECOGNITION_ALGORITHM = 1

paviro, 2016-11-17:
Not sure to be honest not used the tester in a long time... just try to use the module directly there you only need your name :)

Labzz, 2016-11-17:
Already fixed it, the correct line should be: 

personen = ['jd', 'negative', 'othernames', 'vincent']

But now i'm recognized as "othernames" . Its time to go into the real business :D Thanks for the help anyways.

paviro, 2016-11-20:
This should also be fixed now :)

closed 2016-11-20 by paviro