paviro/MMM-Facial-Recognition-Tools #9: cant run python capture.py [Indentation Error]

ItsBradMorgan, 2016-11-16:
IndentationError: expected an indented block
[image]
`
Any ideas?

paviro, 2016-11-17:
Have you changed something in the webcam.py?

ItsBradMorgan, 2016-11-17:
Not at all, and I just double checked...I am running this off the pi via SSH...would that make a difference?

Labzz, 2016-11-17:
@ItsBradMorgan i had the same issue, i made a very easy workaround, with a VERY BASIC script because im lazy. Give it a try;

go to any folder and run : `nano capture.sh`

after paste the following between the lines: 
------------------------------------------------------

#!/bin/bash
mkdir /home/pi/vincent
fswebcam -r 600x800 --no-banner /home/pi/vincent/1.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/2.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/3.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/4.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/5.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/6.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/7.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/8.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/9.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/10.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/11.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/12.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/13.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/14.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/15.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/16.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/17.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/18.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/19.jpg
fswebcam -r 640x480 --no-banner /home/pi/vincent/20.jpg
--------------------------------------------

change the name "vincent" and resolution to your preferences. (not required)
Save it, and run `chmod +x capture.sh`

start the shell script using `sh capture.sh` command and just smile for a while. 

When it's done run the capture.py agains and choose option 2, with the name you filled into the variable earlier. 
Also make sure you choose the right location. Hope it works.

closed 2016-11-20 by paviro