paviro/MMM-Facial-Recognition-Tools #11: facerecognition.py camera.read() crash

Patrick-Remy, 2016-12-28:
It seems like `camera.read()` isn't able to read images. It results in 0x0 images which cause the `cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)` to crash with:

```
Loading training data...
ALGORITHM: LBPH
Training data loaded!
--------------------
Picam ausgewählt...
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 45, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor
```

I am using the Raspberry Pi 3 with v2 NoIR camera. Facetrainer does capturing well.

paviro, 2016-12-28:
Thanks for the investigation I will look into this as soon as possible!

Patrick-Remy, 2016-12-28:
No problem. By the way, the code in MMM-Facial-Recognition is working 👍

paviro, 2016-12-28:
Oh did not even recognize this is only in the tester :D we'll I will see :)

randomstring, 2017-01-02:
I fixed this issue with the following pull request:
https://github.com/paviro/MMM-Facial-Recognition-Tools/pull/12

there's a missing "camera.start()" in facerecognition_tester/lib/config.py to start the camera thread

closed 2017-01-10 by paviro