paviro/MMM-Facial-Recognition-Tools #14: problem adding these Facial Recognition tools and camera module together  in Magic Mirror as a module

vidyaasagar, 2017-03-05:
Hi! i have used these tools and working well individually but i want to include these tools and camera module together in my Magic Mirror project not able to make it work any possibility of making them work. please help me.thanks

Patrick-Remy, 2017-03-09:
Have a look at https://github.com/paviro/MMM-Facial-Recognition

akhil145, 2017-03-31:
Hi! I have cloned modules into modules folder. But while capturing the image for training_data it's showing an error. I installed opencv as given in dependencies using (sudo apt-get install libopencv-dev python-opencv). I enabled the camera in 'raspi-config' but it says it is off and after pressing 'Enter' for capturing image it displays the error like this...

$ python capture.py
What do you want to do?
[1] Capture training images from webcam
[2] Convert '*.jpg' pictures from other cameras to training images
--> 1

Enter the name of the person you want to capture or convert images for.
--> AKHIL
Images will be placed in ./training_data/AKHIL

--------------------
Starting process...

Picam selected...
Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/MagicMirror/modules/tools/facetrainer/lib/picam.py", line 24, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

Capturing image...
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "capture.py", line 35, in <module>
    capture.capture()
  File "/home/pi/MagicMirror/modules/tools/facetrainer/lib/capture.py", line 57, in capture
    image = cv2.cvtColor(image, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

Help me how to resolve this error

closed 2017-04-02 by paviro