paviro/MMM-Facial-Recognition-Tools #15: 2 Camera screens?

Leliekevertje, 2017-03-14:
After having installed the facetrainer and taking some photo's of me, I've tried the face recognition Test tool. When I run python facerecognition.py I get two camera screens. One in a window (with the recognition testing tool) and one screen which is shown in the center of my monitor. I can't get rid of this screen, it stays in front of all other windows. Even Alt-Tab doesn't work. Perhabs it is my fault, but I don't know where I went wrong so please help.

randomstring, 2017-03-14:
The two screens are a feature. One is the picamera ( http://picamera.readthedocs.io/en/release-1.13/api_camera.html  look at the start_preview() method ) and the other is the result of the facial recognition tools and wrapped in the normal window decoration showing the results of the face recognition. The native picamera image is being drawn straight onto the display and you cannot move it around using the window manager. I find that annoying too. You can disable that by commenting out the ```start_preview()``` lines.

https://github.com/paviro/MMM-Facial-Recognition-Tools/search?utf8=%E2%9C%93&q=start_preview

My fork https://github.com/randomstring/MMM-Facial-Recognition-Tools/  adds a config option to disable the picamera preview.

Leliekevertje, 2017-03-15:
Hi Randomstring,

Thanks for the help, it works great!!!

closed 2017-03-15 by Leliekevertje