paviro/MMM-Facial-Recognition-Tools #16: Error while training pictures..

akhil145, 2017-03-31:
Hi! I have cloned modules into modules folder. But while capturing the image for training_data it's showing an error. I installed opencv as given in dependencies using (sudo apt-get install libopencv-dev python-opencv). I enabled the camera in 'raspi-config' but it says it is off and after pressing 'Enter' for capturing image it displays the error like this...

$ python capture.py
What do you want to do?
[1] Capture training images from webcam
[2] Convert '*.jpg' pictures from other cameras to training images
--> 1

Enter the name of the person you want to capture or convert images for.
--> AKHIL
Images will be placed in ./training_data/AKHIL

--------------------
Starting process...

Picam selected...
Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/MagicMirror/modules/tools/facetrainer/lib/picam.py", line 24, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

Capturing image...
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "capture.py", line 35, in <module>
    capture.capture()
  File "/home/pi/MagicMirror/modules/tools/facetrainer/lib/capture.py", line 57, in capture
    image = cv2.cvtColor(image, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

Help me how to resolve this error

randomstring, 2017-03-31:
That error is what you get when the capture() command returns an undefined value. This means that the capture failed. Either the capture.start() didn't happen, the picamera isn't working, OR the picamera didn't warm up in time to start returning images.

I would first test your picamera setup with one of the picamera command line tools to make sure your camera is setup correctly. https://www.raspberrypi.org/documentation/usage/camera/raspicam/raspistill.md

If that works then try adding a ```time.sleep(2)``` after the ```capture.start()``` calls, and see if that fixes it. https://github.com/paviro/MMM-Facial-Recognition-Tools/search?utf8=%E2%9C%93&q=%22capture.start%22&type=

akhil145, 2017-04-01:
I am using raspbian jessie. Since my camera is ps3 eye camera, i can't get it to work while taking pictures. It is working fine while using in motion but it is not taking pictures. when i use "raspistill -o cam.jpg", it is displaying like this..

pi@raspberrypi:~ $ raspistill -o cam.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

i need a solution for this. thankyou

randomstring, 2017-04-01:
Then it's definitely a camera setup problem and not with this module. I did a few google searches for 
https://www.google.com/search?q=mmal%3A+Cannot+read+camera+info%2C+keeping+the+defaults+for+OV5647

and I found the following that suggests this error might be due to multiple programs trying to use the camera at once. Try some of the suggestions from:

http://forum.modmypi.com/technical-support/problem-with-new-pi-camera-t649.html
http://raspberrypi.stackexchange.com/questions/51322/multi-cam-module-error

If you still have problems, you'll be better off posting to a RaspberryPi Camera forum.

closed 2017-04-04 by randomstring