paviro/MMM-Facial-Recognition-Tools #18: Cryptic error in capture.py

martinschaef, 2017-04-16:
When I run `capture.py` on my pi3 with the old pi camera, I get a cryptic error:

```
python capture.py 
['Martin', 'Vivi']
Using FACE_ALGORITM: 1
What do you want to do?
[1] Capture training images from webcam
[2] Convert '*.jpg' pictures from other cameras to training images
--> 1

Enter the name of the person you want to capture or convert images for.
--> Martin
Images will be placed in ./training_data/Martin

--------------------
Starting process...

Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.

Traceback (most recent call last):
  File "capture.py", line 38, in <module>
    capture.capture(preview)
  File "/home/pi/git/MMM-Facial-Recognition-Tools/lib/capture.py", line 54, in capture
    input()
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing
```

Any idea what that could be? (btw, awesome project)

randomstring, 2017-04-16:
input() should probably be raw_input() see this stackoverflow for details:

https://stackoverflow.com/questions/5025920/why-input-gives-an-error-when-i-just-press-enter

Are you using Python 2 or 3? There are some major changes in how input() behaves from 2 ->3. I started down the road of making MMM-Facial-Recognition-Tools python 2 & 3 compatible. However at this time it will only run on Python 2. The input/raw_input issue is where I ran out of steam.

martinschaef, 2017-04-16:
Actually, I'm on 2:
```
python --version
Python 2.7.9
```
but, as you said, changing the line to `raw_input` does the job. Many Thanks!

closed 2017-04-16 by martinschaef