paviro/MMM-Facial-Recognition-Tools #19: Only detects first face in list

martinschaef, 2017-04-22:
On a Pi3, I run into the following problem:
I have two faces, Alice and Bob, with around 30 positive pics each.
I use algorithm 1 and export:
```
export FACE_USERS=Alice,Bob
```
Now if either Alice or Bob stand in front of the camera, it will always detect Alice (with low confidence if its actually Bob). If I export
```
export FACE_USERS=Bob,Alice
```
it always detects Bob. Any idea what might go wrong here?

martinschaef, 2017-04-23:
This is the problem:
https://github.com/paviro/MMM-Facial-Recognition-Tools/blob/master/lib/config.py#L148

That always return a value <= 0.

martinschaef, 2017-04-23:
https://github.com/paviro/MMM-Facial-Recognition-Tools/pull/20

randomstring, 2017-04-23:
I'm having a "what was I thinking" moment. Yeah, that's a bug alright.

martinschaef, 2017-04-23:
max would do here ;)

randomstring, 2017-04-23:
Yes, ```max``` would fix it. Or just replacing it with  ```i - 1``` 

@martinschaef would you be so kind as to create a pull request?

martinschaef, 2017-04-23:
There is one already :)

closed 2017-04-23 by martinschaef