paviro/MMM-Facial-Recognition-Tools #21: Face Recognition TypeError: 'int' object is not iterable

b00002561, 2017-04-24:
Hi, I seem to be having an issue when trying to implement the face recognition tester. When I run the tester on the file I am getting an error, I cannot seem to resolve it.

This is what displays in terminal:

```
Loading training data...

ALGORITHM: LBPH

Training data loaded!

Picam selected...

Traceback (most recent call last):

File "facerecognition.py", line 61, in <module>

label, confidence = model.predict(crop)

TypeError: 'int' object is not iterable
```

My picamera stays active but I can see that it has stopped on my face in the background, Do you have any idea how to resolve this issue, as I am at a loss here.

randomstring, 2017-04-24:
What version of OpenCV are you using? I'm guessing that model.predict() is returning something unexpected.

b00002561, 2017-04-24:
Hi, thanks for the reply I am on opencv 3.1.0, would this be the issue, It was a lot of effort to install opencv took me a few days to get it installed so am hesitant to try change the version

randomstring, 2017-04-24:
Support for openCV 3 is new and experimental. You're probably the first to try to get it to work. The issue is that the functional interface for the face recognizer has changed from CV 2 -> CV 3.  All that has been done so far is detect opencv3 and instantiate the model object correctly:

https://github.com/paviro/MMM-Facial-Recognition-Tools/blob/master/lib/config.py#L118

I would encourage you to find the other changes and submit a pull request once you've fixed them all. :)

FYI. Here is my personal mini-HOWTO for installing OpenCV. Maybe it will save you some time.
https://github.com/randomstring/learn-python/blob/master/opecv_pi_install.md

b00002561, 2017-04-24:
Thanks a million for your help, would you recommend that I install the older version of opencv?
Do I need to uninstall the new version? Sorry for all the questions, I am learning this as I go. Not sure If I would be qualified to submit a pull request yet haha

randomstring, 2017-04-24:
Installing OpenCV is a nightmare. Especially if you google it, all the top results have you install from source. Don't do it. I have wasted DAYS trying to install openCV from source. Use apt-get (on pi or linux) and live with the default version. If you're reverting from CV3 to CV2, you should nuke all the install directories and install CV2 form a package.

b00002561, 2017-04-24:
Sorry, I feel stupid for asking but I am only getting to know linux too, so I need to uninstall opencv 3, not sure how to do that, and then I need to download opencv2 and install that? Is that correct?

randomstring, 2017-04-24:
If you installed from source, you might be able to run ```make uninstall``` in the source directory. If you're lucky, the OpenCV distribution builders added an uninstall option and it works.

Otherwise you'll have to find what directory that the OpenCV 3 library files were installed and delete the files by hand.

You might get away with installing openCV 2 and ignore openCV 3, but that's a recipe for future problems with something linking to the wrong version. The resulting errors can be mysterious and misleading.

b00002561, 2017-04-24:
Ok perfect, and last question, can I just use the code you put in earlier to install the right version of opencv?

randomstring, 2017-04-24:
The following aren't really instructions, they're a reminder to my future self how I got openCV to install painlessly. Big assumptions in this command sequence is the specific version of python (2.7), the path to the python libraries, and the use of virtualenvwrapper for tracking various python & pip installations (cv_py2 is a name I chose for a set of packages I have installed). So take care to make sure you tailor the ```workon`` and ```cp``` commands to your installation. If you don't use ```virtualenvwrapper``` or ```virtualenv``` you only really need the first command to ```apt-get``` the packages.

```% sudo apt-get install libopencv-dev python-opencv
% cp /usr/lib/python2.7/dist-packages/cv* ~/.virtualenvs/cv_py2/lib/python2.7/site-packages/
% workon cv_py2
% python -V
Python 2.7.9
% python
>>> import cv2
>>> cv2.__version__
'2.4.9.1'
>>> quit()```

b00002561, 2017-04-25:
HI, I installed OpenCV 2.4.10 and have checked its working, 

I then went to the facial recognition tester again and ran the programme, I received the following error

```
pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition/facerecognition_tester$ python facerecognition.py 
Loading training data...
ALGORITHM: LBPH
Training data loaded!
Picam selected...
Traceback (most recent call last):
  File "facerecognition.py", line 67, in <module>
    cv2.putText(frame, config.users[label - 1], (x - 3, y - 8), cv2.FONT_HERSHEY_SIMPLEX, 1.5, (255, 255, 255), 1)
IndexError: list index out of range
```
Do you know how to fix this? Should I delete the whole module and start again?

randomstring, 2017-04-25:
Are on the latest?  That doesn't error line doesn't match the most current code in https://github.com/paviro/MMM-Facial-Recognition-Tools/blob/master/facerecognition.py

A ```git pull``` should fix it.

b00002561, 2017-04-25:
HI, I have done a git pull but receive the same error, this is not the main module, I am trying to get the facial recognition _tester to work? Not sure if that would make a difference?

closed 2017-05-03 by paviro