paviro/MMM-Facial-Recognition-Tools #22: capture.py failed to take picture

MyProgTools, 2017-04-26:
Hy,

when i run capture.py I've got the following error, 
Does anyone have an idea?

geert@PI3:~/MagicMirror/modules/MMM-Facial-Recognition-Tools-master$ python capture.py
Remember to set the name list environment variable FACE_USERS
Using default FACE_ALGORITM: 1
What do you want to do?
[1] Capture training images from webcam
[2] Convert '*.jpg' pictures from other cameras to training images
--> 1

Enter the name of the person you want to capture or convert images for.
--> Geert
Images will be placed in ./training_data/Geert

--------------------
Starting process...

Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.

Traceback (most recent call last):
  File "capture.py", line 38, in <module>
    capture.capture(preview)
  File "/home/geert/MagicMirror/modules/MMM-Facial-Recognition-Tools-master/lib/capture.py", line 54, in capture
    input()
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

davidnesbitt7, 2017-04-27:
Having the exact same error, although I have File "<string>", line 0 at the bottom, not sure if it's an SSH problem or not, it's setting up my folder with my name in it but no images being pulled over, I'm seeing the preview of the image on my screen just when it goes to take the image it's spazzing out?!

davidnesbitt7, 2017-04-27:
Solved it, I read back through the old issues and it had been solved sorry, hopefully it helps you too @MyProgTools in the /lib/ folder open the capture.py file in the text editor/terminal, find input() and replace it with raw_input(), then save the file and run as instructed

MyProgTools, 2017-04-28:
Thanks, i will change line 24 and line 54 to raw_input() this evening.

MyProgTools, 2017-04-28:
It works. :-) Thank you. Now I can go on.
I had already reviewed the old issues, but I find nothing with this error message.

davidnesbitt7, 2017-04-28:
Just realised I should have linked you to it too, it's in issue #18 someone linked to [this](https://stackoverflow.com/questions/5025920/why-input-gives-an-error-when-i-just-press-enter) and it explained that for python 2 - input() works fine, but for python 3 - raw_input() is needed

closed 2017-05-13 by MyProgTools