paviro/MMM-Facial-Recognition-Tools #25: Xlib:  extension "RANDR" missing on display ":10.0". Missing OpenGL

skippengs, 2017-05-31:
I cannot run this line:
`pip install -r requirements.txt`

I get the following error:
`Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
Storing debug log for failure in /home/pi/.pip/pip.log
`

I can capture and train with the camera. But whenever I run 
`python facerecognition.py`

I get the following error:
`pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ export FACE_ALGORITHM=1`
`pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ export FACE_USERS=Skippy`
`pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ python facerecognition.py`
`['Skippy']`
`Using FACE_ALGORITM: 1`
`Loading training data...`
`Training data loaded!`
`Xlib:  extension "RANDR" missing on display ":10.0".`
`(Facial recognition:3900): GdkGLExt-WARNING **: Window system doesn't support OpenGL.`

When it runs, the preview pops-up on the hdmi connected screen. Whenever it is done loading the the training data it immediately crashes.

Could this be because i cannot run the required.txt python line? Or is something else missing?
Kind of new to these things and i have reread the tutorials multiple times and tried to google extensively.

randomstring, 2017-05-31:
Looks like requirements.txt didn't get checked in. :/

In the meantime run ```pip install future``` by hand.

skippengs, 2017-05-31:
I forgot to mention that I did install future by hand. So that's not it.

randomstring, 2017-05-31:
Based on the error message I'm guessing either 1) you aren't running this with an active X display (either via a remote login or from the console) or 2) some OpenGL problem? 

The ``facerecognition.py`` script opens a window to display an image of the resulting labelled face.

skippengs, 2017-05-31:
I am logging in with Remote Desktop. But the pi has a screen connected to it. Could this really be it?

closed 2017-05-31 by skippengs