paviro/MMM-Facial-Recognition-Tools #28: Traceback when capture.py

atimon, 2017-07-25:
Hello,
I am trying to run:
`python capture.py` in pi@raspberrypi:~/MMM-Facial-Recognition-Tools and i get the following error:
```
Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    import lib.capture as capture
  File "/home/pi/MMM-Facial-Recognition-Tools/lib/capture.py", line 12, in <module>
    from   builtins import input
ImportError: No module named builtins
```
Is there a specific place where i need to git clone this tool? what am i doing wrong? I am currently trying to set this up with a magicmirror.
thanks.

peglegsqueeks, 2017-08-04:
Hi there, 
I am having the exact same issue, I'm sure it's probably something I missed.

Any help would be greatly appreciated as I'm very new to github.

Thank you

peglegsqueeks, 2017-08-04:
**Solution**: try this installs missing dependancies. It worked for me! 

`sudo pip install future`

Cheers

closed 2017-08-24 by atimon