paviro/MMM-Facial-Recognition-Tools #30: Camera is not detected

atimon, 2017-08-24:
I am using a USB camera and when I try to run `python facerecognition.py` I get this:
```
Loading training data...
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/MMM-Facial-Recognition-Tools/lib/picam.py", line 27, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

Training data loaded!
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "facerecognition.py", line 58, in <module>
    image = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor
```
any help?

mathsha, 2017-09-14:
I'm having a similar error with my USB webcam too.

```
Starting process...

Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/picam.py", line 27, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
```

Can anyone help?

seattlecajun, 2018-01-07:
I am having the same problem with my USB camera not being detected. I've installed fswebcam and confirmed that the camera is usable to the pi as that app is able to collect and save an image from the USB webcam.

When I try running python capture.py, I get the following:

Capturing positive training images.
Press enter to capture an image.
Press Ctrl-C to quit.
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/picam.py", line 27, in run
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

Capturing image...
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/imgproc/src/color.cpp, line 3737
Traceback (most recent call last):
  File "capture.py", line 41, in <module>
    capture.capture(preview)
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/capture.py", line 60, in capture
    image = cv2.cvtColor(image, cv2.COLOR_RGB2GRAY)
cv2.error: /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/imgproc/src/color.cpp:3737: error: (-215) scn == 3 || scn == 4 in function cvtColor

randomstring, 2018-01-07:
This is not a facial recognition error, this is a raspberry pi camera setup issue. Try this:

https://www.raspberrypi.org/documentation/usage/webcams/

seattlecajun, 2018-01-08:
Thanks for getting back to me. I had already reviewed the article you mentioned and installed/tested fswebcam to take snapshots with my USB webcam - that part is all working.

In the interim, I tested with an rpi cam (non-usb) and the module worked flawlessly - truly outstanding actually. Now that I've removed the rpi-cam, I'm trying to get the USB webcam working.  I edited the config file to "useUSBCam: true" and disabled the camera interface in raspi-config.

Now, when I startup MagicMirror, the USB webcam's LEDs light up as if the camera is active but recognition no longer works. I get the following after starting up MM:

`Starting MagicMirror: v2.2.2
Loading config ...
Loading module helpers ...
Initializing new module helper ...
Module helper loaded: MMM-Facial-Recognition
Initializing new module helper ...
Module helper loaded: MMM-Remote-Control
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: MMM-GoogleMapsTraffic.
No helper found for module: compliments.
Initializing new module helper ...
Module helper loaded: MMM-MyWeather
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ...
You're using a full whitelist configuration to allow for all IPs
Server started ...
Connecting socket for: MMM-Facial-Recognition
Starting module helper: MMM-Facial-Recognition
Connecting socket for: MMM-Remote-Control
Starting node helper for: MMM-Remote-Control
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: MMM-MyWeather
MMM-WunderGround helper started ...
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Launching application.
ERROR! Could not find main module js file for MMM-Facial-Recognition-Tools
Create new calendar fetcher for url: https://calendar.google.com/calendar/ical/marc.spears%40gmail.com/private-bec9cfdb4ca9479537513218042e7af9/basic.ics - Interval: 300000
Create new calendar fetcher for url: http://www.facebook.com/ical/b.php?uid=833007719&key=AQD5LcwIE-OiBsRN - Interval: 300000
GET_WUNDERGROUND
Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
Create new news fetcher for url: http://feeds.bbci.co.uk/news/world/rss.xml - Interval: 300000
Create new news fetcher for url: http://www.thehindu.com/news/national/?service=rss - Interval: 300000
Create new news fetcher for url: http://kuow.org/feeds/term/393/rss.xml - Interval: 300000
[MMM-Facial-Recognition] Facerecognition started...
[MMM-Facial-Recognition] Loading training data...
[MMM-Facial-Recognition] ALGORITHM: LBPH
[MMM-Facial-Recognition] Training data loaded!
[MMM-Facial-Recognition] --------------------
[MMM-Facial-Recognition] Webcam ausgewählt...
`

seattlecajun, 2018-01-12:
Just wanted to update this for others having issues with USB cameras. I followed the advice of another user on the [MM forums ](https://forum.magicmirror.builders/topic/210/facial-recognition-customize-your-mirror-for-every-user/200?page=20) and fixed the issue that I was having. 

In my case, after MMM-Facial-Recognition detected that I didn't have an rpi camera, it wasn't 'failing over' to the usb webcam. What worked for me was editing the module's config.py file to comment out the rpi camera entirely; taking it out of the equation, as it were. 

Open up `/MagicMirror/modules/MMM-Facial-Recognition/lib/config.py` and comment out the below lines:

> > def get_camera(preview=True):
>     try:
> #        from . import picam
> #        capture = picam.OpenCVCapture(preview)
> #        capture.start()
> #        return capture
>         from . import webcam
>         return webcam.OpenCVCapture(device_id=0)
>         capture.start()
>         return capture
>     except Exception:
>         from . import webcam
>         return webcam.OpenCVCapture(device_id=0)

I realize editing the module's config file directly isn't really a good long-term solution but it did the trick for me.

marktillo, 2018-05-15:
The solution I found was to swap import picam to webcam and swap capture=picam to webcam in the config.py file then it never looks for the picam at all