paviro/MMM-Facial-Recognition-Tools #31: dont understand the "usage" part

kaisern07, 2017-12-17:
[image]

Hi,i dont understand the third and fourth part... is it ask us to find FACE_ALGORITHM and FACE_USERS in the config.py and change the following variables to "1" and "user name we set earlier" ?  anyone to help will be appreciated

randomstring, 2017-12-17:
Steps 3 and 4 are for adding environment variables to your unix shell. You just run the `export` commands from the same command line as you run ` python facerecognition.py`

This is so you don't have to edit the python code and change the defaults in the code. The default `FACE_ALGORITHM` is probably the best. If you don't set the `FACE_USERS` I think the code defaults to using `USER1`, `USER2`, etc.

For more information on how to set and read shell environment variables, I found this good tutorial:
https://www.digitalocean.com/community/tutorials/how-to-read-and-set-environmental-and-shell-variables-on-a-linux-vps

randomstring, 2017-12-17:
A quick google search indicates that yes you can set shell environment variables on Windows 10. If you're familiar with python, you may be better off just editing the code to set those two variables. Good luck.

kaisern07, 2017-12-18:
Alright...I will give it a try ... Thank you so answering