paviro/MMM-Facial-Recognition-Tools #32: ValueError in capture.py

seattlecajun, 2018-01-10:
In an attempt to fix a prior issue (not able to use my USB webcam), I installed the full version of OpenCV following instructions here: `https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/`

Unfortunately, that seems to have caused another issue. When I run capture.py now, it's giving me an error about needing more than 3 values to unpack.

```
pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ python capture.py
Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    import lib.capture as capture
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/capture.py", line 21, in <module>
    from . import config
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/config.py", line 15, in <module>
    (CV_MAJOR_VER, CV_MINOR_VER, mv1, mv2) = cv2.__version__.split(".")
ValueError: need more than 3 values to unpack
```
When I enter python on its own and run the following commands, here is the output that I get:

```
pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> (CV_MAJOR_VER, CV_MINOR_VER, mv1, mv2) = cv2.__version__.split(".")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'cv2' is not defined
>>> import inspect
>>> import os
>>> import platform
>>> import cv2
>>> (CV_MAJOR_VER, CV_MINOR_VER, mv1, mv2) = cv2.__version__.split(".")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: need more than 3 values to unpack
```

I know just enough to get myself into more trouble! :) Any advice that could help me fix this?  Thanks!

kaisern07, 2018-01-11:
this code is not compatible to opencv 3, try use opencv2

seattlecajun, 2018-01-12:
Closing the issue as I ended up re-installing the OS and MagicMirror. Thanks. :)

closed 2018-01-12 by seattlecajun