paviro/MMM-Facial-Recognition-Tools #33: No module named buildins

seattlecajun, 2018-01-12:
Hi there. I recently re-installed Raspbian Stretch and Magic Mirror then reconfigured most of the modules I was using before. I'm having a problem with this one again though.

When I attempt to run `python capture.py`, I get the following:

> pi@raspberrypi:~/MagicMirror/modules/MMM-Facial-Recognition-Tools $ python capture.py
> Traceback (most recent call last):
>   File "capture.py", line 18, in <module>
>     import lib.capture as capture
>   File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/capture.py", line 12, in <module>
>     from   builtins import input
> ImportError: No module named builtins

As stated this is a fresh install of the mirror and I've installed the opencv dependencies. Anyone have an idea of what I'm doing wrong here? Thanks!

randomstring, 2018-01-12:
run `pip install future`

I just added the missing `requirements.txt` with the pip future module in it.

seattlecajun, 2018-01-12:
Thanks @randomstring, I found that mentioned on the MM forums earlier tonight and tried it out - it fixed this issue completely!

closed 2018-01-12 by seattlecajun