paviro/MMM-Facial-Recognition-Tools #34: No Module Named Builtins

BluishMechanismz, 2018-03-22:
Hi. I've been having the same problem others have been having, and even after installing future, the exact same problem continues. Reboots and reinstalls don't help either.

`Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    import lib.capture as capture
  File "/home/pi/MMM-Facial-Recognition-Tools/lib/capture.py", line 12, in <module>
    from   builtins import input
ImportError: No module named builtins`

randomstring, 2018-03-23:
If you installed the future module with pip, then I suspect you may have some issue with the version of python you're running not matching the environment your pip modules are getting installed into.

MuhammadJamee, 2018-09-02:
Same problem for me, is there any solution?

horstefan, 2018-10-30:
Same problem. any updates on this?

aniiketdongare07, 2020-03-20:
Same issue, any updates?

randomstring, 2020-03-21:
You all probably need to install the python 2 future package. Try:

`pip install future`

closed 2020-03-21 by randomstring