paviro/MMM-Facial-Recognition-Tools #37: train.py 'createLBPHFaceRecognizer' error

rubinho101, 2020-03-15:
train.py not running on **MacOS 10.11.6 & OCV 3.3.0.10** 
----------

I tried also `LBPHFaceRecognizer_create` instead of `createLBPHFaceRecognizer` with no success.

I use ocv 3.3.0.10 because other versions dont seem to work on my macos 10.11.6

`('Read', 6, 'positive images and', 400, 'negative images.')
6 images from subject xy`

`Training model type 1 with threshold 80
Traceback (most recent call last):
  File "train.py", line 115, in <module>
    model = config.model(config.RECOGNITION_ALGORITHM, config.POSITIVE_THRESHOLD)
  File "/Users/ruben/MMM-Facial-Recognition-Tools/lib/config.py", line 135, in model
    model = cv2.createLBPHFaceRecognizer(threshold=thresh)
AttributeError: 'module' object has no attribute 'createLBPHFaceRecognizer'`

**AttributeError: 'module' object has no attribute 'createLBPHFaceRecognizer'**

aniiketdongare07, 2020-03-21:
I got the same error, can you pls tell me if you got ahead of this?

randomstring, 2020-03-21:
This is probably a problem with your OpenCV install. OpenCV can be a real pain to install.

I found the best way to install OpenCV on the mac is using brew.

Also, LBPHFaceRecognizer kinds sucks at recognizing faces. The reason there's has been no active work on this module in over 2 years. I spent some time fixing it up, only to realize it just doesn't work.

closed 2020-03-21 by randomstring