paviro/MMM-Facial-Recognition-Tools #39: errors running capture.py

bert269, 2020-12-24:
Followed the instructions, All pre-req's are installed: When I run capture.py I get these messages:
python capture.py
Traceback (most recent call last):
  File "capture.py", line 18, in <module>
    import lib.capture as capture
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/capture.py", line 21, in <module>
    from . import config
  File "/home/pi/MagicMirror/modules/MMM-Facial-Recognition-Tools/lib/config.py", line 15, in <module>
    (CV_MAJOR_VER, CV_MINOR_VER, mv1, mv2) = cv2.__version__.split(".")
ValueError: need more than 3 values to unpack

What am I missing?

randomstring, 2020-12-25:
Honestly, this code doesn't work well and there are many better face recognition platforms than the OpenCV module that this uses. This modules is no longer supported see https://github.com/paviro/MMM-Facial-Recognition-Tools#depreciated

If you insist on trying this code... I would guess that the OpenCV version is not what the code expects. You could print out cv2.version and debug from there

randomstring, 2020-12-25:
maybe this pull request fixes it?

https://github.com/paviro/MMM-Facial-Recognition-Tools/pull/26